Tysonpower/HCPBridgeMqtt_tynet #6: Consequences of "taking control" in ESPHome

sNiXx, 2025-02-14:
Could you maybe describe what the consequences of clicking the "Take Control" button are? The ESPHome UI mentions a few aspects, but I guess these are not the only changes? Is there maybe a more in-depth explanation in the documentation of ESPHome?

[image]

Also, how would one go about changing the WiFi configuration? For instance, if I wanted to switch to a different SSID or change the password? Would I have to connect the board via USB-C again?

Thank you :)

PS: The integration is working perfectly, great work!

Tysonpower, 2025-02-15:
This is exactly hat the take control feature in ESPHome is for :)

The ESPHome firmware is preconfigured so it's easy to setup, you can than take control of the device yourself to change it's configuration and update things like wifi ssid, api encryption etc. directly in the esphome config it creates in the UI of your HA ESphome Plugin.

This is a standard ESPHome feature.

closed 2025-02-15 by Tysonpower

sNiXx, 2025-02-22:
When I took control, I ran into a few errors, because the following secrets were missing:
- hcp_wifi_ap_password
- api_key
- web_username
- web_password

I guess it is obvious when looking at [esphome.yaml](https://github.com/Tysonpower/HCPBridgeMqtt_tynet/blob/main/esphome.yaml), but it would definitely help if this was in the documentation. I could create another PR if you want.

Also, I am seeing some errors and warnings regarding the BME280 sensor in the logs. Should I just ignore them?

<details><summary>Error Details</summary>

```
INFO ESPHome 2024.12.4
INFO Reading configuration /config/esphome/hcpbridge.yaml...
INFO Updating https://github.com/Tysonpower/esphome-hcpbridge.git@None
INFO Starting log output from 192.168.0.53 using esphome API
INFO Successfully connected to hcpbridge @ 192.168.0.53 in 0.036s
INFO Successful handshake with hcpbridge @ 192.168.0.53 in 0.076s
[23:15:23][I][app:100]: ESPHome version 2024.12.4 compiled on Feb 22 2025, 22:32:45
[23:15:23][I][app:102]: Project tynet.hcpbridge_e4 version 1.0.0
[23:15:23][C][wifi:600]: WiFi:
[23:15:23][C][wifi:428]:   Local MAC: [redacted]
[23:15:23][C][wifi:433]:   SSID: [redacted]
[23:15:23][C][wifi:436]:   IP Address: 192.168.0.53
[23:15:23][C][wifi:440]:   BSSID: [redacted]
[23:15:23][C][wifi:441]:   Hostname: 'hcpbridge'
[23:15:23][C][wifi:443]:   Signal strength: -56 dB ▂▄▆█
[23:15:23][C][wifi:447]:   Channel: 1
[23:15:23][C][wifi:448]:   Subnet: 255.255.255.0
[23:15:23][C][wifi:449]:   Gateway: 192.168.0.1
[23:15:23][C][wifi:450]:   DNS1: 192.168.0.1
[23:15:23][C][wifi:451]:   DNS2: 0.0.0.0
[23:15:23][C][logger:185]: Logger:
[23:15:23][C][logger:186]:   Level: DEBUG
[23:15:23][C][logger:188]:   Log Baud Rate: 9600
[23:15:23][C][logger:189]:   Hardware UART: USB_CDC
[23:15:23][C][i2c.arduino:071]: I2C Bus:
[23:15:23][C][i2c.arduino:072]:   SDA Pin: GPIO6
[23:15:23][C][i2c.arduino:073]:   SCL Pin: GPIO5
[23:15:23][C][i2c.arduino:074]:   Frequency: 50000 Hz
[23:15:23][C][i2c.arduino:086]:   Recovery: bus successfully recovered
[23:15:23][C][light:092]: Light 'Garage Door Light'
[23:15:23][C][hcpbridge.binary_sensor:033]: HCPBridgeIsConnected
[23:15:23][C][hcpbridge.binary_sensor:019]: HCPBridgeRelaySensor
[23:15:23][C][hcpbridge.binary_output:018]: HCPBridgeBinaryOutput
[23:15:23][C][hcpbridge.light:043]: HCPBridgeBinaryLight
[23:15:23][C][bme280_i2c.sensor:025]:   Address: 0x76
[23:15:23][C][bme280.sensor:182]: BME280:
[23:15:23][E][bme280.sensor:185]: Communication with BME280 failed!
[23:15:23][C][bme280.sensor:194]:   IIR Filter: OFF
[23:15:23][C][bme280.sensor:195]:   Update Interval: 30.0s
[23:15:23][C][bme280.sensor:197]:   Temperature 'BME280 Temperature'
[23:15:23][C][bme280.sensor:197]:     Device Class: 'temperature'
[23:15:23][C][bme280.sensor:197]:     State Class: 'measurement'
[23:15:23][C][bme280.sensor:197]:     Unit of Measurement: '°C'
[23:15:23][C][bme280.sensor:197]:     Accuracy Decimals: 1
[23:15:23][C][bme280.sensor:198]:     Oversampling: 16x
[23:15:23][C][bme280.sensor:199]:   Pressure 'BME280 Pressure'
[23:15:23][C][bme280.sensor:199]:     Device Class: 'pressure'
[23:15:23][C][bme280.sensor:199]:     State Class: 'measurement'
[23:15:23][C][bme280.sensor:199]:     Unit of Measurement: 'hPa'
[23:15:23][C][bme280.sensor:199]:     Accuracy Decimals: 1
[23:15:23][C][bme280.sensor:200]:     Oversampling: 16x
[23:15:23][C][bme280.sensor:201]:   Humidity 'BME280 Humidity'
[23:15:23][C][bme280.sensor:201]:     Device Class: 'humidity'
[23:15:23][C][bme280.sensor:201]:     State Class: 'measurement'
[23:15:23][C][bme280.sensor:201]:     Unit of Measurement: '%'
[23:15:23][C][bme280.sensor:201]:     Accuracy Decimals: 1
[23:15:23][C][bme280.sensor:202]:     Oversampling: 16x
[... 42 lines cut ...]

```
i2c:
  sda: GPIO06
  scl: GPIO05
  scan: false

sensor:
  - platform: bme280_i2c
    temperature:
      name: "BME280 Temperature"
    pressure:
      name: "BME280 Pressure"
    humidity:
      name: "BME280 Humidity"
    address: 0x76 # can be 0x77 as well
    update_interval: 30s
  - platform: dht
    pin: GPIO04
    temperature:
      name: "DHT Temperature"
    humidity:
      name: "DHT Humidity"
    update_interval: 30s
```

Tysonpower, 2025-02-22:
The errors in the logs can and should be ignored, if they annoy you feel free to comment out or remove the sensor definitions of BMW and DHT if you don have them connected. 

The reason the sensors are in the default config is to avoid people having to change anything when they add one of those sensors, what is really common.

For Documentation: feel free to make a PR, happy to get help with that, also you maybe better pointing out what is missing from your standpoint as a user.

sNiXx, 2025-02-23:
Added information to the documentation in #8

Tysonpower, 2025-02-23:
Merged, thanks for the work!