Tysonpower/HCPBridgeMqtt_tynet #10: Webinterface enhancements

MartinMeise, 2025-02-26:
In the Webinterface i would suggest to replace the text in the buttons with Symbols:
up:       ↑
Stop:   ◯
Down:  ↓
I would position this three on the left or right side of the picture of the door one below the other. Like most standard Garage Door buttons are arranged.
The once for fixt positions like half or vent can remain below the door.

Tysonpower, 2025-03-03:
Great idea, the UI is not used that much tho directly, but sounds like a simple thing to do, i will have a look if i can implement this soon.

Tysonpower, 2025-03-03:
did some work on the UI, i guess this is what you had in mind?

[image]

MartinMeise, 2025-03-03:
Yes, looks very good. Thank you.

Tysonpower, 2025-03-03:
okay will make an updated ota image in the following days for it

Tysonpower, 2025-03-18:
New Firmware 0.8.0 now has the WebUI fixes implemented

closed 2025-03-18 by Tysonpower