Tysonpower/HCPBridgeMqtt_tynet #9: Controlling a shally directly

MartinMeise, 2025-02-26:
I'm using this board to control a garage door. I also have a light which is controlled via a Shally 1pm mini. It would be great if I could set up a URL which is called when the door is open and another one when it is closed. Or I could write a script like only switch on the light at night.
MQTT would probably be an option but as I have not set up a MQTT nor a Homa Assistant yet a direct possibility would be nice. A direct function would also work if I would use the Wifi Access point function of the Shally.

Tysonpower, 2025-03-03:
This is a very special function i think, not sure if this is something that is useful for many people tho.

Are both devices in the same lan?

For sure you can always fork and modify the firmware yourself and make a PR, it is doable for sure, but not sure if i will implement it.

MartinMeise, 2025-03-03:
Yes, both are in the same LAN. It's a simple URL requests.
Good to know that I can do a frog. Maybe I'll have look in to that. But I only did some smaller arduino, Java Script or Delphi projects so far. In the Shelly Webinterface you could wright Java Script to do some actions.

14yannick, 2025-03-19:
In Esphome would just be a couple of lines in the yaml.
https://esphome.io/components/http_request.html