Tysonpower/HCPBridgeMqtt_tynet #12: Recommendation for additional sensors?

manuschillerdev, 2025-03-23:
Thanks for the module - it works absolutely great!
The readme states, that the module would support additional sensors

> DS18X20 temperature sensor
BME280 temperature and humidity sensor
DHT22 temperature and humidity sensor
HC-SR04 proximity sensor (car (below) detection)
HC-SR501 PIR motion sensor

Are there recommendations for specific modules that are known to be working out of the box with this RS485 Bridge? I am new to IOT and sometimes a little lost when it comes to buying compatible hardware tbh 

Thanks!

Tysonpower, 2025-03-23:
Hi there, happy to hear that it works great, there is a doc for DHT22 and BME280 - you can just get a "5V BME280" module like in my photos and it will work out of the box with the ESPhome version, the factory firmware needs a OTA Update (prebuild bins are avaliable)

https://github.com/Tysonpower/HCPBridgeMqtt_tynet/blob/main/docs/connecting_sensors_prebuilt_pcbs.md

Duke8123, 2025-03-26:
In the photo's it says it should be a 3v BMP module. The 3v modules I can find are with 6 pins. The 5v version is with 4 pins, like in the photo's. Should it be a 5v BMP module instead, like you state here?

Tysonpower, 2025-03-26:
Both sensor work, the 5V version works great even on 3V3 what the PCB outputs on the EXP_1 Port.

So if you get either the 5V or the 3V3 Version, both will do the job.

Tysonpower, 2025-03-26:
For Supramatic 3 you will use the ESPHome Firmware that comes preinstalled, with that there is no need for an software update, the firmware should recognize the sensor. If not you can adapt the device in ESPHome and change the Config if needed. Example yaml: 

https://github.com/Tysonpower/HCPBridgeMqtt_tynet/blob/main/esphome_e3.yaml

Tysonpower, 2025-03-26:
Pls send me an email to support@tynet.eu - include some screenshots from the device info in home assistant, thanks :)

Tysonpower, 2025-03-28:
Issue of @cousinhub57 was fixed, e3 yaml is missing the sensor confg, will update that later today in the repo

closed 2025-07-12 by manuschillerdev