Tysonpower/HCPBridgeMqtt_tynet #14: BLE Proxy / ESP-IDF

sNiXx, 2025-04-10:
Hi @Tysonpower, I was just reading a bit on the BLE proxy functionality of ESPHome and was wondering if you had tested any of this? My use case is that I would like to use a Shelly BLU Button 1 to open my garage door and the device would be in a perfect location to use as a BLE proxy.

If I read the documentation at [Bluetooth Proxy](https://esphome.io/components/bluetooth_proxy) correctly, I think a few changes would be necessary, because:
- _The ESP32 Platform component should be configured to use the esp-idf framework, as the arduino framework uses significantly more memory and performs poorly with the Bluetooth proxy enabled._ 
- _The ESP-IDF framework [...] is recommended for variants of the ESP32 like ESP32S2, ESP32S3, ESP32C3 and single-core ESP32 chips._
- _The Web Server Component component should be disabled as the device is likely to run out of memory and will malfunction when both components are enabled simultaneously._

Of course, the device should be partitioned due to differences between the Arduino and ESP-IDF framework.

``` yaml
packages:
  tynet.hcpbridge_e4: github://Tysonpower/HCPBridgeMqtt_tynet/esphome.yaml@main

esp32:
  board: adafruit_feather_esp32s3
  framework:
    type: esp-idf

web_server: !remove

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true
```

Tysonpower, 2025-04-10:
Nice idea, did you try to just compile and test it like that? If so, did it work?

sNiXx, 2025-04-11:
No, I was a bit hesitant, as I am still quite new to ESP32/ESPHome and was worried I could brick my device while partitioning 😆

Can you think of any negative side effects? As far as I can tell from the documentation and some additional research, the ESP-IDF framework should be mainly beneficial, as it removes the Arduino layer (which itself seems to be using ESP-IDF underneath), so that would definitely be something I would be interested in as well.

Tysonpower, 2025-04-11:
You can' really brick an ESP32S3 like on my PCBs, so don't worry.

I think ESP-IDF in ESPHome has some issues with some libraries, but it should work, Arduino does use esp-idf in the background that's correct :)

sNiXx, 2025-04-12:
I tried to compile it today, but received an error regarding a missing Arduino dependency, which kind of makes sense to me. However, I am not quite sure what the solution would be.

Also, I was thinking of giving the BLE proxy a try without ESP-IDF, but how would I perceive "poor performance", as mentioned in the docs?

<details>
<summary>hcpbridge.yaml</summary>

```
packages:
  tynet.hcpbridge_e4: github://Tysonpower/HCPBridgeMqtt_tynet/esphome.yaml@main

esp32:
  board: adafruit_feather_esp32s3
  framework:
    type: esp-idf

web_server: !remove

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true
```

</details>

<details>
<summary>logs</summary>

```
In file included from src/esphome/components/hcpbridge/button/../hcpbridge.h:9,
                 from src/esphome/components/hcpbridge/button/hcpbridge_button.h:5,
                 from src/esphome/components/hcpbridge/button/hcpbridge_button.cpp:1:
src/esphome/components/hcpbridge/button/../hoermann.h:7:10: fatal error: Arduino.h: No such file or directory

*****************************************************************
* Looking for Arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Arduino.h"
* Web  > https://registry.platformio.org/search?q=header:Arduino.h
*
*****************************************************************

    7 | #include <Arduino.h>
      |          ^~~~~~~~~~~
compilation terminated.
*** [.pioenvs/hcpbridge/src/esphome/components/hcpbridge/button/hcpbridge_button.cpp.o] Error 1
========================= [FAILED] Took 440.23 seconds =========================
```

</details>

Tysonpower, 2025-04-13:
looks like the hörmann code used the arduino lib, i will try some things in the next days and let you know

sNiXx, 2025-04-13:
That would be much appreciated, thank you 😃

I just attempted to compile for the Arduino framework (I'm a bit curious, but also impatient 😜 ) and ran into an error regarding program size.

<details>
<summary>hcpbridge.yaml</summary>

```
packages:
  tynet.hcpbridge_e4: github://Tysonpower/HCPBridgeMqtt_tynet/esphome.yaml@main

web_server: !remove

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true
```

</details>

<details>
<summary>logs</summary>

```Error: The program size (1510665 bytes) is greater than maximum allowed (1441792 bytes)
*** [checkprogsize] Explicit exit, status 1
RAM:   [==        ]  21.0% (used 68836 bytes from 327680 bytes)
Flash: [==========]  104.8% (used 1510665 bytes from 1441792 bytes)
========================= [FAILED] Took 292.16 seconds =========================
```

</details>

sNiXx, 2025-04-25:
I was able to reduce the size just enough by removing many additional parts from the yaml file. However, before I install this firmware on my device: Were you able to figure something out regarding Arduino.h?

<details>
<summary>hcpbridge.yaml</summary>

```
packages:
  tynet.hcpbridge_e4: github://Tysonpower/HCPBridgeMqtt_tynet/esphome.yaml@main

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true

# adjustments to save space
sensor: !remove
captive_portal: !remove
i2c: !remove
improv_serial: !remove
web_server: !remove
logger:
  level: INFO
```

</details>

Tysonpower, 2025-04-28:
Just try it, i wasn't able to have a look on fixing the code or porting it to esp idf without arduino.h so far due to TIme.

Tysonpower, 2025-09-30:
My current board that i ship use a ESP32-S3-WROOM-1-N8 - is that enough or is the R8 version needed?

Of course you could just replace the chip on the board :)

mdklapwijk, 2025-10-01:
Sorry I went by the images on the site; still need to remove my board from the hormann to try sNiXx' yaml, so I'll have a look later today.

manuschillerdev, 2025-10-07:
Am I understanding that correctly that in combination with a ble beacon I could automate Opening and closing the garage door? :)

manuschillerdev, 2025-10-19:
> Guess what? This seems to work perfectly; the first time you need to flash via usb for esphome to be able to repartition flash:
> ```
> esp32:
>   board: adafruit_feather_esp32s3
>   flash_size: 8MB
> 
> esp32_ble_tracker:
>   scan_parameters:
>     active: false
> 
> bluetooth_proxy:
>   active: true
> ```
> 
> I am actually running it with custom interval/window parameters, due to testing [this](https://github.com/agittins/bermuda/issues/668):
> ```
> esp32_ble_tracker:
>   scan_parameters:
>     interval: 1100ms
>     window: 1100ms
>     active: false
> ```
> 
> As for the esp-idf question; I had a brief look, but that is quite some work, as the component modbus-esp8266 needs to be replaced by esp-modbus for all ModbusRTU-slave functions used.
> 
> update: the door seems to react a bit slower sometimes, so trying 320 interval and 120 window next.

@mdklapwijk sorry - no expert on this. Could You please give me a quick how-to for enabling ble Tracking on the stock tynet hörmann board?

Tysonpower, 2025-11-20:
> I can’t get it running but im also inexperienced. My .yaml in ESPhome Device Builder, running as an AddOn in Home Assistant, looks like the one above, but it compiles with an error:
> 
> ```
> INFO ESPHome 2025.11.0
> INFO Reading configuration /config/esphome/hcpbridge.yaml...
> Failed config
> 
> packages: [source /config/esphome/hcpbridge.yaml:2]
>   
>   Failed to load packages. esphome.yaml is not a valid YAML file. Please check the file contents.
>   Secret 'api_key' not defined
>     in "/data/packages/ae96e053/esphome.yaml", line 142, column 10.
>   tynet.hcpbridge_e4: github://Tysonpower/HCPBridgeMqtt_tynet/esphome.yaml@main
> ```
> 
> I entered my API and WiFi credentials. I try to do it wireless.

The Error message says exactrly what you are missing, the api key secret. See example here under config point 6.

https://github.com/Tysonpower/HCPBridgeMqtt_tynet/blob/main/docs/getting_started_esphome.md#alternative-setup-procedure

mdklapwijk, 2025-12-04:
Ok, I missed this commit f75fa42139416cd1a1fbc3555e651544c8925bb9, so the esp32 part should use the correct variant instead of the feather board definition:
```
...
esp32:
  variant: esp32s2
  flash_size: 8MB
...
```