Tysonpower/HCPBridgeMqtt_tynet #15: Device ID not updating - Conflict in MQTT communication

pjator, 2025-04-23:
**Describe the bug**
I have two units and all works fine with one connected at a time. But when both are connected one will stop working. 

**To Reproduce**
I have connected and disconnected both separately to ensure it is linked to that both are connected at the same time.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**

[image]

[image]

[image]

[image]

**Additional context**
When I look in the sysinfo (and also in my router data) I can see that the Hostname is still HCPBRIDGE so changes to device ID has not changed the hostname.

pjator, 2025-04-23:
I managed to find a workaround by changing the MQTT users to be different, this way they function without the mismatch. But I assume the bug is still relevant. But at least working now with this workaround, but would of course prefer if the same MQTT user could be used for both entities.

Tysonpower, 2025-04-28:
Hi there, didn't see the github issue sorry. The DeviceID and Hostname are two different things, the device ID in MQTT should change so MQTT can see both devices and the Hostname can stay the same. This is how many people are using it currently.

How does your MQTT Info look? Do they really report the same deviceID in MQTT Explorer or similar?

pjator, 2025-04-28:
No they do get different Device IDs in the MQTT broker. But they are somehow getting confused and I lose connection in Home Assistant with one of the entities. 

So I am not sure the real reason for the issue is caused by the Hostname, and with two different MQTT users it works fine. 

So you can close the issue if you want but, it is a fact that the hostname does not get changed even if you try to change it. 

Best regards
P

Tysonpower, 2025-04-28:
WIll do some tests when i find time and will keep the issue open for now. Do you see any issues in the MQTT or HA logs when using the same MQTT user? What type of MQTT server do you use?

pjator, 2025-04-29:
I am using Mosquitto broker version 6.5.0

Can't really find any meaningfull error in the log, just this portion that happens when both are connected at the same time with the same user (mqtt). 

> `2025-04-23 22:09:00: New client connected from 10.10.10.251:64543 as ESP-EC07364AEC24 (p2, c1, k15, u'mqtt').
2025-04-23 22:09:00: New connection from 10.10.10.237:58308 on port 1883.
2025-04-23 22:09:00: New client connected from 10.10.10.237:58308 as ESP-BC08364AEC24 (p2, c1, k15, u'mqtt').
2025-04-23 22:09:00: Client ESP-EC07364AEC24 disconnected due to malformed packet.
2025-04-23 22:09:00: Client ESP-BC08364AEC24 disconnected due to malformed packet.`

And now when I have connected with two users mqtt1 and mqtt2 it works: 

> `2025-04-28 14:12:04: New client connected from 10.10.10.251:56766 as ESP-EC07364AEC24 (p2, c1, k15, u'mqtt2').
2025-04-28 14:12:15: New connection from 10.10.10.237:54220 on port 1883.
2025-04-28 14:12:15: New client connected from 10.10.10.237:54220 as ESP-BC08364AEC24 (p2, c1, k15, u'mqtt1').
2025-04-28 14:26:24: Saving in-memory database to /data//mosquitto.db.`

So as mentioned I am not sure if the hostname has anything to do with the issue, it is the only thing I can find that sticks out that I can't change the hostname as seen the ID here are different when it connects. But I get this malformed packet issue. 

Best regards
Peter

Tysonpower, 2025-09-07:
Please test again with new version 0.8.3

Tysonpower, 2025-12-10:
hcpbridgedoor1 is to long i think, numbers do work but thanks for the information!

closed 2026-04-05 by Tysonpower