Tysonpower/HCPBridgeMqtt_tynet #16: One device dropping out

rtho782, 2025-05-04:
**Describe the bug**
I have two doors each with one of these RS485 bridges purchased from your store on 5th Jul 2024. One of them has started dropping off from the network. When it does, it shows as still connected in Unifi and on the DHCP scope of my Mikrotik router, but won't respond to pings, webui won't load, etc. Lights stay on on the device.

Pulling the connection out and plugging it back in several times brings it back online, for a few hours.

I can access the OTA update screen, but cannot find updated firmware.

The 2nd device (I assume same version, can't see fw version listed) is fine.

The device has been working for 10 months, so not sure what has changed!!

Tysonpower, 2025-05-06:
Try this firmware file: https://github.com/Tysonpower/HCPBridgeMqtt_tynet/blob/main/HCPBridgeESP32/fw/esp32s3_tynet.bin

Or you can switch to the ESPHome Firmware: https://github.com/Tysonpower/HCPBridgeMqtt_tynet/blob/main/docs/getting_started_esphome.md

Please send me an email to support@tynet.eu if you need further help, i will read that way quicker.

Tysonpower, 2025-05-24:
In contact over email, should be resolved.

closed 2025-05-24 by Tysonpower