Tysonpower/HCPBridgeMqtt_tynet #25: Update to ESPHome 2025.7.1 - AsyncTCP.cpp:14:10: fatal error: NetworkInterface.h: No such file or directory

t-nebel, 2025-07-17:
**Describe the bug**
After updating to ESPHome 2025.7.1 (from 2025.6.X), an update of the esp (ESP32-S3 RS485 Bridge from tynet shop) is not possible.
Maybe some outdated dependencies regarding AsyncTCP?!

**To Reproduce**
Just clicking "update" in ESPHome UI. Same on Install Wireless.

Log:
```
INFO ESPHome 2025.7.1
INFO Reading configuration /config/esphome/eggaragehoermannbridge.yaml...
INFO Updating https://github.com/Tysonpower/esphome-hcpbridge.git@None
INFO Generating C++ source...
INFO Compiling app...
Processing eg-garage-hoermannbridge (board: adafruit_feather_esp32s3; framework: arduino; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip)
--------------------------------------------------------------------------------
Library Manager: Installing ESP32Async/AsyncTCP @ 3.4.5
INFO Installing ESP32Async/AsyncTCP @ 3.4.5
Unpacking  [####################################]  100%
Library Manager: AsyncTCP@3.4.5 has been installed!
INFO AsyncTCP@3.4.5 has been installed!
Library Manager: Installing emelianov/modbus-esp8266
INFO Installing emelianov/modbus-esp8266
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: modbus-esp8266@4.1.0 has been installed!
INFO modbus-esp8266@4.1.0 has been installed!
Library Manager: Installing ESP32Async/ESPAsyncWebServer @ 3.7.10
INFO Installing ESP32Async/ESPAsyncWebServer @ 3.7.10
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer@3.7.10 has been installed!
INFO ESPAsyncWebServer@3.7.10 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing bblanchon/ArduinoJson @ 7.4.2
INFO Installing bblanchon/ArduinoJson @ 7.4.2
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@7.4.2 has been installed!
INFO ArduinoJson@7.4.2 has been installed!
Library Manager: Installing improv/Improv @ 1.2.4
INFO Installing improv/Improv @ 1.2.4
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: Improv@1.2.4 has been installed!
INFO Improv@1.2.4 has been installed!
Library Manager: Installing esphome/noise-c @ 0.1.10
INFO Installing esphome/noise-c @ 0.1.10
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.10 has been installed!
INFO noise-c@0.1.10 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10020.7
INFO Installing esphome/libsodium @ 1.10020.7
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10020.7 has been installed!
INFO libsodium@1.10020.7 has been installed!
HARDWARE: ESP32S3 240MHz, 320KB RAM, 4MB Flash
 - framework-arduinoespressif32 @ 3.1.3 
 - framework-arduinoespressif32-libs @ 5.3.0+sha.489d7a2b3a 
 - tool-esptoolpy @ 4.8.6 
 - tool-mklittlefs @ 3.2.0 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-riscv32-esp @ 13.2.0+20240530 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Dependency Graph
|-- Networking @ 3.1.3
|-- AsyncTCP @ 3.4.5
|-- modbus-esp8266 @ 4.1.0
[... 102 lines cut ...]
Compiling .pioenvs/eg-garage-hoermannbridge/lib4c5/Network/NetworkManager.cpp.o
Compiling .pioenvs/eg-garage-hoermannbridge/lib4c5/Network/NetworkServer.cpp.o
Compiling .pioenvs/eg-garage-hoermannbridge/lib4c5/Network/NetworkUdp.cpp.o
Compiling .pioenvs/eg-garage-hoermannbridge/libca5/AsyncTCP/AsyncTCP.cpp.o
Archiving .pioenvs/eg-garage-hoermannbridge/lib4c5/libNetwork.a
Compiling .pioenvs/eg-garage-hoermannbridge/lib718/modbus-esp8266/Modbus.cpp.o
.piolibdeps/eg-garage-hoermannbridge/AsyncTCP/src/AsyncTCP.cpp:14:10: fatal error: NetworkInterface.h: No such file or directory

**************************************************************************
* Looking for NetworkInterface.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:NetworkInterface.h"
* Web  > https://registry.platformio.org/search?q=header:NetworkInterface.h
*
**************************************************************************

   14 | #include <NetworkInterface.h>
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/eg-garage-hoermannbridge/libca5/AsyncTCP/AsyncTCP.cpp.o] Error 1
========================= [FAILED] Took 120.21 seconds =========================
```

**Expected behavior**
Update of the esp as usual ;)

MaWeSX, 2025-07-17:
same issue here, was also present for 2025.6.2 to 2025.7.0

Thanks for having a look

closed 2025-07-18 by Tysonpower

Tysonpower, 2025-07-18:
Fixed the yaml, a flag caused the issue, please make sure you remove the deep+ flag under the platformio_options.

Validation should look something like that if it got the new yaml, don't get fooled by a cache :)

[image]

t-nebel, 2025-07-19:
Great, works again like a charm - thank you for the support!

commit 5fd5a4b7 (2025-07-18): removing lib_ldf_mode: deep+

Fixes #25 - for some reason the lib_idf_mode has issues resolving AsyncTDPs networkInterface.h file.
commit 61b7ac3e (2025-07-18): Update esphome_e3.yaml to fix #25