Tysonpower/HCPBridgeMqtt_tynet #26: Door will close when button "open" is pressed AND if door is already open

Olgrov, 2025-07-18:
In case the Garage Door is already fully open (100%) and when you press in Home Assistant the button "open", then the Garage Door will close [ESPHome Firmware, commands via Home Assistant].
Same happens when you set the opening position to 100% via Home Assistant. If the door is lower than 100%, it will go up to 100%. But in case the door was already 100% open, the door will shut down to 0%. This might be dangerous, because you would not expect the door to close when you send the "open" command.

In case the door is just below 100% (e.g. 60% open), then the open command will open as intended. In case the door is currently opening or closing, then the open command will stop the door (which is also as expected) - same behaviour as the remote control.

Here an excerpt of the debug logfile:

Status: Garage door whas fully open (100%) when open command was sent. Then the garage door automatically closes.

15:38:04 [D] [hcpbridge.cover:114] HCPBridgeCover::update() - state changed
15:38:04 [D] [cover:170] 'Garage Door' - Publishing:
15:38:04 [D] [cover:173] Position: 100%
15:38:04 [D] [cover:186] Current Operation: IDLE
15:38:04 [D] [hcpbridge.text_sensor:052] HCPBridgeTextSensor::update() - Open
15:38:04 [D] [text_sensor:064] 'Garage Door State': Sending state 'Open'
15:38:04 [W] [component:237] Component hcpbridge took a long time for an operation (65 ms).
15:38:04 [W] [component:238] Components should block for at most 30 ms.

[command "open" pressed in home assistant

15:38:07 [D] [cover:076] 'Garage Door' - Setting
15:38:07 [D] [cover:084] Position: 100%
15:38:08 [D] [binary_sensor:036] 'Garage Door Relay state': Sending state ON
15:38:11 [D] [binary_sensor:036] 'Garage Door Relay state': Sending state OFF
15:38:11 [D] [hcpbridge.cover:114] HCPBridgeCover::update() - state changed
15:38:11 [D] [cover:170] 'Garage Door' - Publishing:
15:38:11 [D] [cover:173] Position: 99%
15:38:11 [D] [cover:186] Current Operation: CLOSING
15:38:11 [D] [hcpbridge.text_sensor:052] HCPBridgeTextSensor::update() - Closing
15:38:11 [D] [text_sensor:064] 'Garage Door State': Sending state 'Closing'
15:38:11 [W] [component:237] Component hcpbridge took a long time for an operation (57 ms).
15:38:11 [W] [component:238] Components should block for at most 30 ms.
15:38:12 [D] [hcpbridge.cover:114] HCPBridgeCover::update() - state changed
15:38:12 [D] [cover:170] 'Garage Door' - Publishing:
15:38:12 [D] [cover:173] Position: 97%
15:38:12 [D] [cover:186] Current Operation: CLOSING
15:38:12 [D] [hcpbridge.cover:114] HCPBridgeCover::update() - state changed
15:38:12 [D] [cover:170] 'Garage Door' - Publishing:
15:38:12 [D] [cover:173] Position: 94%
15:38:12 [D] [cover:186] Current Operation: CLOSING
15:38:13 [D] [cover:076] 'Garage Door' - Setting
15:38:13 [D] [cover:080] Command: STOP
15:38:13 [D] [hcpbridge.cover:114] HCPBridgeCover::update() - state changed
15:38:13 [D] [cover:170] 'Garage Door' - Publishing:
15:38:13 [D] [cover:173] Position: 91%
15:38:13 [D] [cover:186] Current Operation: CLOSING
15:38:13 [D] [hcpbridge.cover:114] HCPBridgeCover::update() - state changed
15:38:13 [D] [cover:170] 'Garage Door' - Publishing:
15:38:13 [D] [cover:173] Position: 90%
15:38:13 [D] [cover:186] Current Operation: IDLE
15:38:13 [D] [hcpbridge.text_sensor:052] HCPBridgeTextSensor::update() - Stopped
15:38:13 [D] [text_sensor:064] 'Garage Door State': Sending state 'Stopped'
15:38:13 [W] [component:237] Component hcpbridge took a long time for an operation (63 ms).
15:38:13 [W] [component:238] Components should block for at most 30 ms.

Tysonpower, 2025-07-18:
never had this issue, what motor do you have?

Olgrov, 2025-07-18:
> never had this issue, what motor do you have?

SupraMatic E4

This only happens when the command comes from Home Assistant. If you press the "open" command via direct access to the device via it's ip-adress, nothing will happen (which is good, because the door is already open).

Tysonpower, 2025-07-18:
How do you even press the open button in HA? For me it's grey and not clickable

[image]

Olgrov, 2025-07-18:
> How do you even press the open button in HA? For me it's grey and not clickable
> 
> <img alt="Image" width="570" height="612" src="https://private-user-images.githubusercontent.com/13482963/468074889-439a92a8-a461-4117-8568-f4089b152090.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTI4NDgyNTAsIm5iZiI6MTc1Mjg0Nzk1MCwicGF0aCI6Ii8xMzQ4Mjk2My80NjgwNzQ4ODktNDM5YTkyYTgtYTQ2MS00MTE3LTg1NjgtZjQwODliMTUyMDkwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzE4VDE0MTIzMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBjNWUzZGMyNjY4YWFhNzI2MjA3NjEyMWYxNjI5NGE1ODYyYWJjNDY3NjhmZmM2ZWI2YzA3YjQ1ZmJjNmU1NzgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.4jhSyb05rt5GGXi7pE7OEMNjIyRnmC-BoNP99Ze_5Aw">

For me its not grey, unfortunately. I can click it:

[image]

Olgrov, 2025-07-18:
Btw. wenn the door is already closed and you press the close button (button is also not greyed out), nothing will happen.
--> Logfile:
16:39:26 [D] [cover:076] 'Garage Door' - Setting
16:39:26 [D] [cover:084] Position: 0%

Nothing else is reported in the logs.

 So the bug only exists when trying to open an already open door.

Olgrov, 2025-07-18:
What's also weird:
Via direct access of the device by it's IP-Address, the Button "open" only works when the door is fully closed. The highlighted button has no function in case the door is more than 0% open.

[image]

In addition: the Relay state also has a strange behaviour.
Sometimes it is on, sometimes it isn't.

When the bug appears (open an open door), then the logfiles tell the following about the relay:
[Pressing open when garage is already open]
17:08:32 [D] [cover:076] 'Garage Door' - Setting
17:08:32 [D] [cover:084] Position: 100%
17:08:32 [D] [binary_sensor:036] 'Garage Door Relay state': Sending state ON
17:08:35 [D] [hcpbridge.cover:114] HCPBridgeCover::update() - state changed
17:08:35 [D] [cover:170] 'Garage Door' - Publishing:
17:08:35 [D] [cover:173] Position: 100%
17:08:35 [D] [cover:186] Current Operation: CLOSING
17:08:35 [D] [binary_sensor:036] 'Garage Door Relay state': Sending state OFF
17:08:35 [D] [hcpbridge.text_sensor:052] HCPBridgeTextSensor::update() - Closing

Tysonpower, 2025-07-18:
only idea i have is that the firmware you flashed ( so my file because the build fails) is not compatible with the current HA version. After #25 is fixed, please flash with current firmware and test again

Tysonpower, 2025-07-18:
Compile issue is fixed, please try again with current yaml in this repo.