Tysonpower/HCPBridgeMqtt_tynet #33: Very Simple (?) Feature Request

deggle, 2025-09-01:
It would be really great if we could expose a command to simply 'toggle' or 'step' the door state. It's obviously great having proper states for open/close etc, however I've a few instances that I want to map a single button to work like a basic style open/stop/close/stop input.

I could implement this logic externally but it would make so much more sense to add it into here.

I think it would just need a new HA_STEP string setting up and a few lines of code in onMqttMessage to pick up the command and do the state change based on the current state. I would be open to doing it, but I think it's a small change that for someone familiar with the code would take 2 minutes...

Many thanks, Tim

Tysonpower, 2025-09-02:
It's not only 2mins of work, but a quick change for sure.
have a look if this logic is what you are looking for - should be right i guess.

[image]

If you send me an email to support@tynet.eu i can send you a beta version for you to test as well.

deggle, 2025-09-03:
Hello, thank you for looking at this!

The issue here is that unless fully closed (or open) the movement state needs to alternate to give a open-stop-close-stop-open... type toggling on the command (may want to name it 'STEP' rather than toggle, as I think that's what it would be called on the garage door instructions - it's the way a single button remote works). 

So, I think a variable will be needed to store the _last_ movement state, so the logic is a but more like:
If closed then open
If opening||closing then stop, but also set lastState =currState
If stopped & lastState = opening then close
Else If stopped & lastState = closing then open

Does that make sense?

Kind regards,

Tim

Tysonpower, 2025-09-03:
Yeah makes sense, i guess it should be correct now:

[image]

deggle, 2025-09-03:
Looks right to me!
I'll drop you an email.
Thanks!

deggle, 2025-09-06:
Sorry for the delay, thanks for the binary. I'm not able to get this working - I'm sending "step" via MQTT to [prefix]/command/door and nothing happens. Sending open/close/stop to this topic works as expected.

Tysonpower, 2025-09-06:
Let me tinker in the next days with it, will get back to you

Tysonpower, 2025-09-07:
Added this feature under Release 0.8.3

closed 2025-09-07 by Tysonpower